bit-bots/bitbots_motion #50: refactor spline engine and stabilizer

SammyRamone, 2019-07-30:
we decided that we want to refactor our splines in a way that we have an abstract spline engine to get a common interface and to have utility functions (like debug markers for all splines) in one place.
related to this is also small refactoring of the spline classes (e.g. adding debug marker methods).
The stabilizer will also be an abstract class. Implementation will maybe be based on existing libraries (e.g. ros controllers).
The engine and stabilizer should than be used in QuinticWalk, DynamicKick and DynUp

SammyRamone, 2019-07-30:
as a library for controller, the base library which is also used in ros control would be a simple candidate https://github.com/ros-controls/control_toolbox

ftsell, 2019-08-29:
@timonegk can you upload the notes we took when we discussed this?

ftsell, 2019-09-09:
I've received them from Timon: [splines.txt](https://github.com/bit-bots/bitbots_motion/files/3590487/splines.txt)

ftsell, 2019-09-09:
I've also recently stumbled upon [Behavior Driven Development](https://cucumber.io/docs/bdd/) and would like to try it out. I think the implementation corresponding to this issue might be a good trial.

timonegk, 2019-10-31:
This was closed in #63, #64 and #89.

closed 2019-10-31 by timonegk