bit-bots/bitbots_motion #85: animation walkready is not always exactly reached

ftsell, 2019-10-02:
This is a bug appearing to be in the animation server and affecting hcm.

It's manifestation is a robot which is in the walkready position but keeps twitching. In this case hcm is stuck in the playing walkready state.

My educated guess is that the Robot does not reach the _exact_ walkready position for some reason causing the animation server to not finish the _play amimation_ action in turn causing the hcm to be stuck.

My proposed resolution is to have an allowed error margin in the animation server. Inside that margin an animation would be considered _completed_ (but maybe only after a timeout).

SammyRamone, 2019-10-04:
did you do it on the robot or in simulation?
the animation server plays the animation open loop, so its not possible that it gets stuck there.
The HCM has an error margin, which may needs to be increased, especially for simulation, since the knees seem to bend more there

ftsell, 2019-10-05:
I reported this error while @NFiedler was working on the robot.

If the AnimationServer works open loop, that is indeed probably not the problem.

But I believe we had this error in Sydney at some point as well. @timonegk or @SammyRamone don't you remember?

SammyRamone, 2019-11-07:
I don't remember if there was a problem in sydney with it. If the bug is still present, we can just increase the error margin in the HCM config. otherwise we can close the issue

timonegk, 2019-11-07:
The problem persists and increasing the error margin is not sufficient because the robot is not stable in the reached position. The problem might be that the motors (especially the ankle pitches) do not reach their target positions.

SammyRamone, 2019-11-07:
> The problem persists and increasing the error margin is not sufficient because the robot is not stable in the reached position. The problem might be that the motors (especially the ankle pitches) do not reach their target positions.

but why does increasing the error margin does not help? it is specially for this case, where the actual position of the robot differs from the animation file

timonegk, 2020-02-19:
Maybe the walkready animation is restarted before the previous execution is finished.

SammyRamone, 2020-08-25:
does not happen anymore on new robot version

closed 2020-08-25 by SammyRamone