bit-bots/bitbots_motion #129: Odometry during stand up is wrong

timonegk, 2020-02-06:
## Expected behavior
During the stand up, the odometry should still be correct.

## Current behavior
The odometry is wrong, the robot seems to be in the ground.

## Possible Solution
This might be due to the odometry fuser that selects the roll and pitch of the IMU and the yaw of the walking which might be wrong when the robot is lying.

[image]

Flova, 2020-02-08:
The weird rotation is due to the gimbal lock at 90° Pitch. This can be fixed by using a quaternion for the merging process, instead of the RPY coordinates.

Flova, 2020-02-08:
I think this would be helpfull, to determin the Rotation around the Z Axis.

Flova, 2020-02-19:
Closed in #132

closed 2020-02-19 by Flova