bit-bots/bitbots_motion #208: HCM: load animations in the beginning

SammyRamone, 2021-02-17:
Currently the HCM is loading the falling animations everytime they are played. This requires always reading again the animation file, which propably takes time. This is not a good idea if we want to act fast during falling.

## Describe the solution you'd like
Load all relevant animation files in during init once and then use the data that is in the RAM

SammyRamone, 2021-03-25:
we should also look at the animation server and see if it has the same issue

SammyRamone, 2021-03-25:
actually, the HCM does not do this but only the animation server. in the animation_node.py the get_animation_splines method reads the animations from the file. Instead all animations should be read in the init and saved as some kind of dict.

ftsell, 2021-03-25:
Just so you know, Linux saves recently accessed files in a cache and transparently serves them from RAM.

But this is a transparent, non-controlled, unreliable solution for us so I do support this issue. I'm simply saying that it might not bring the desired performance increase.

SammyRamone, 2021-03-25:
> Just so you know, Linux saves recently accessed files in a cache and transparently serves them from RAM.
> 
> But this is a transparent, non-controlled, unreliable solution for us so I do support this issue. I'm simply saying that it might not bring the desired performance increase.

Good to know. Still, the first time the robot falls it needs time to open the file for the falling animation. Even for this one case it would be worth it.

closed 2021-05-14 by jaagut