bit-bots/bitbots_motion #246: HCM does not set robot state to GETTING_UP

timonegk, 2021-05-24:
## Expected behavior
During the standup motion, the robot state should be GETTING_UP (3).

## Current behavior
The state stays FALLEN (2).

## Steps to Reproduce
1. `roslaunch bitbots_bringup simulator.launch`
2. `rostopic echo /robot_state`
3. Throw the robot in the simulator to the ground and let it stand up.

## Possible Solution
Could be done similar to how it is done for the kick.

SammyRamone, 2021-05-25:
The issue is that there is basically no time that we are actually in the state "FALLEN", since we directly start standing up again. We could change it so that the HCM publishes "GETTING_UP" but then it would basically never publish "FALLEN" and I think some software components depend on it.
Do you have some use case where you actually need to receive the "GETTING_UP"?

SammyRamone, 2021-05-26:
After some discussion, we decided it should just be renamed to getting up in this case since we are never really staying in the fallen state

closed 2021-05-27 by SammyRamone